lazd/VoodooI2CGoodix #15: Does not load correctly on GPD P2 Max

coderobe, 2020-01-16:
## Prerequisites
* [x] I have followed the [VoodooI2C installation instructions](https://voodooi2c.github.io/#Installation/Installation)
* [x] I have [patched my DSDT](https://github.com/alexandred/VoodooI2C-Patches)
* [x] I have followed the [GPIO pinning guide](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning)
* [x] VoodooI2C recognizes my I2C controllers during a verbose boot
* [x] I have followed the [VoodooI2CGoodix installation instructions](https://github.com/lazd/VoodooI2CGoodix#installation)
* [x] I have tried tapping my screen and nothing happens

## Environment
 - **Computer**: GPD P2 Max
 - **Goodix Touchscreen model**: unknown GDIX1002
 - **VoodooI2C version:** 2.3
 - **VoodooI2CGoodix version:** 0.2.4
 - **macOS Version:** 10.15.2 (19C57)

## `DSDT.zip`
[DSDT.aml.zip](https://github.com/lazd/VoodooI2CGoodix/files/4069956/DSDT.aml.zip)

## `IOReg`
[image]

## `VoodooI2C.log`
Empty

## Additional context
When VoodooI2C and VoodooI2CGoodix are injected via Clover, VoodooI2C loads but VoodooI2CGoodix either never loads or is unloaded by the time the system is booted. Loading VoodooI2CGoodix manually using kextutil keeps it loaded, but the situation does not change.

Azkali, 2020-01-21:
Your DSDT file is not patched correctly.
I made a working one inside my [repository](https://github.com/Azkali/GPD-P2-MAX-HACKINTOSH)

Azkali, 2020-01-21:
For a quick fix use VoodooI2CGoodix version: 0.2.3.
Also your DSDT is different than [mine](https://github.com/Azkali/GPD-P2-MAX-HACKINTOSH) maybe that's also related.

Azkali, 2020-01-21:
The driver name is `GDIX1001` inside `VoodooI2CGoodix.kext -> Contents -> Info.plist -> IOKitPersonnalities -> Goodix Touch Screen -> IOPropertyMatch -> name` so changing to `GDIX1002` should solve your issue, no need to rebuild anything just open the file Info.plist inside the kext driver

lazd, 2020-01-22:
@coderobe thanks for the report, I believe I can update the plist to match both `GDIX1001` and `GDIX1002`. I'll do so and perform another release.

lazd, 2020-01-22:
@coderobe give [VoodooI2CGoodix 0.3.0](https://github.com/lazd/VoodooI2CGoodix/releases/tag/v0.3.0)  a try, it should load up no problem!

coderobe, 2020-01-22:
Thanks a lot for looking at this!
It loads up fine now, and stays loaded, but the touchscreen still does not react to me tapping or dragging my finger around on the screen. The system log is not showing anything, still.

@Azkali have you tested this in your repo? I cannot reproduce it, even using your files instead of mine.

lazd, 2020-01-22:
@coderobe hmm interesting. Can you get the logs of it loading now that it's loading properly? That will give us some insight.

My logs from boot don't show up properly due to a misconfiguration in our DSDT, so maybe you're having the same issue... If no logs show up when you running the following:

```
sudo log show --predicate "processID == 0" --last 2m --debug --info  | grep Goodix
```

Then give this a shot:

```
sudo mkdir -p /Volumes/EFI/
sudo mount -t msdos /dev/disk0s1 /Volumes/EFI # assumes disk0 is your boot disk
mkdir -p ~/tmp
cp -r /Volumes/EFI/EFI/CLOVER/kexts/Other/VoodooI2C*.kext ~/tmp/
sudo chown -R root:wheel ~/tmp/VoodooI2C*.kext
rm -rf /Volumes/EFI/EFI/CLOVER/kexts/Other/VoodooI2CGoodix.kext
```

Then restart. Once restarted, try this:

```
sudo kextutil -d ~/tmp/VoodooI2C.kext ~/tmp/VoodooI2CGoodix.kext
sudo log show --predicate "processID == 0" --last 2m --debug --info  | grep Goodix
```

That will load up the kext and dump out all of the logs. Dump that here and we can debug.

coderobe, 2020-01-23:
Ah yes, that does show logs!

```
2020-01-23 07:38:34.400354+0100 0x1397     Default     0x0                  0      0    kernel: (kernel) VoodooI2CGoodixTouchDriver::Probing
2020-01-23 07:38:34.400402+0100 0x1397     Default     0x0                  0      0    kernel: (kernel) VoodooI2CGoodixTouchDriver::Reading version...
2020-01-23 07:38:34.400596+0100 0x1397     Default     0x0                  0      0    kernel: (kernel) VoodooI2CGoodixTouchDriver::Read version failed: -536870212
2020-01-23 07:38:34.400606+0100 0x1397     Default     0x0                  0      0    kernel: (kernel) VoodooI2CGoodixTouchDriver::Failed to init device
2020-01-23 07:38:34.400626+0100 0x1397     Default     0x0                  0      0    kernel: (kernel) VoodooI2CGoodixTouchDriver::Freeing
```

lazd, 2020-01-23:
@coderobe hmm, that's a generic I2C error. I was getting those before we added this to the device in DSDT:

```
Scope (_SB.PCI0.I2C2)
{
    // begin add
    Name (SSCN, Package (0x03)
    {
        0x01B0, 
        0x01FB, 
        0x1E
    })
    Name (FMCN, Package (0x03)
    {
        0x48, 
        0xA0, 
        0x1E
    })
    // end add
    Device (TCSE)
    {
        // ...
    }
}
```

If that doesn't work, try replacing `ActiveLow` with `ActiveHigh` inside of `SBFG`. Take a look at the [MiniBook DSDT](https://github.com/THEDEVIOUS1/CHUWI-MINIBOOK-HACKINTOSH/blob/master/EFI/CLOVER/ACPI/patched/DSDT.aml) and see if you can adapt our changes to make yours work.

In any case, I think this is a DSDT issue, so can you join in the conversation over here https://github.com/Azkali/GPD-P2-MAX-Hackintosh/issues/20 and see if y'all can troubleshoot it together?

lazd, 2020-01-23:
@Azkali thoughts on the above? We had to do a bit more DSDT patching to get I2C to work at all. Looking at your DSDT, it doesn't seem like it would work.

Did you successfully test the touchscreen under Catalina? If so, can you paste the logs so I can see what version of the touchscreen your device has?

coderobe, 2020-01-23:
I decompiled @Azkali's DSDT and merged their relevant differences a few minutes ago and now i am getting further, but interacting with my screen still does nothing
```
VoodooI2CGoodixTouchDriver::Probing
VoodooI2CGoodixTouchDriver::Reading version...
VoodooI2CGoodixTouchDriver::ID 928, version: 1060
VoodooI2CGoodixTouchDriver::Reading config...
VoodooI2CGoodixTouchDriver::xOutputMax = 2560
VoodooI2CGoodixTouchDriver::yOutputMax = 1600
VoodooI2CGoodixTouchDriver::maxTouches = 10
VoodooI2CGoodixTouchDriver::screenTouchLevel = 80
VoodooI2CGoodixTouchDriver::screenLeaveLevel = 60
VoodooI2CGoodixTouchDriver::lowPowerInterval = 3
VoodooI2CGoodixTouchDriver::refreshRate = 1
VoodooI2CGoodixTouchDriver::xThreshold = 0
VoodooI2CGoodixTouchDriver::yThreshold = 0
VoodooI2CGoodixTouchDriver::Device initialized
VoodooI2CGoodixTouchDriver::VoodooI2CGoodixTouchDriver has started
VoodooI2CGoodixEventDriver::Published multitouch interface
VoodooI2CGoodixEventDriver::Got active display
VoodooI2CGoodixEventDriver::Got active framebuffer
VoodooI2CGoodixEventDriver::Configuring multitouch interface with dimensions 2560,1600 and 11 transducers
```

coderobe, 2020-01-23:
Given https://github.com/Azkali/GPD-P2-MAX-Hackintosh/issues/20, i am not confident anyone has gotten it to actually work yet.

lazd, 2020-01-23:
@coderobe that's a good sign! I2C communication is working. I can send you a debug version later to test to see if it's getting any kind of touch events.

You can also try compiling the project yourself and uncommenting the debug defines [here](https://github.com/lazd/VoodooI2CGoodix/blob/master/VoodooI2CGoodix/VoodooI2CGoodixEventDriver.hpp#L44) and [here](https://github.com/lazd/VoodooI2CGoodix/blob/master/VoodooI2CGoodix/VoodooI2CGoodixTouchDriver.hpp#L21). I have some handy [shell functions](https://gist.github.com/lazd/16d89b72c905624cc5d6d37aeff0056a) that can be used to aid in loading/unloading/logging etc.

lazd, 2020-01-23:
Here's a debug version, does it log anything interesting when you poke it? Try the `streamgoodixlog` command from my shell scripts.

[VoodooI2CGoodix-0.3.1-debug.zip](https://github.com/lazd/VoodooI2CGoodix/files/4102068/VoodooI2CGoodix-0.3.1-debug.zip)

coderobe, 2020-01-23:
Using your build there's nothing new in the logs, sadly, so i think it's not actually getting any touch events at all.

lazd, 2020-01-23:
@coderobe sounds like an issue in the interrupt configuration. Perhaps investigate that, and run through the GPIO pinning guide and make sure you didn't miss something.

Azkali, 2020-01-23:
Sorry did not saw your message until now. I do have a working touchscreen with Catalina, I can provide my logs, one thing to note is that I'm using an unlocked BIOS, thus it shouldn't impact the touchscreen.
I do agree with you @lazd I think the DSDT may need further patching

1548976092, 2020-01-23:
I still ca n’t use the touch screen

coderobe, 2020-01-23:
@Azkali can you share your bios and/or other firmware patches you have done that are not available in your repository? I'd like to investigate this further.

Azkali, 2020-01-23:
Surely, the only file altered is my DSDT.aml 
[DSDT.aml.zip](https://github.com/lazd/VoodooI2CGoodix/files/4104264/DSDT.aml.zip)

Azkali, 2020-01-23:
Oh I see why my repository isn't properly working, so basically I'm an idiot .. I'm gonna fix that quickly

coderobe, 2020-01-23:
Your DSDT does not have the power state methods mine has. I have removed those from mine and it appears to be working - though the cursor stutters and drops all events every second or so. I suspect the touchscreen potentially enters some low power state which it is not coming back from.

Azkali, 2020-01-23:
Yes, I also noticed stutters and and dropping.. I've just updated my repo to fix the DSDT issue

lazd, 2020-01-23:
@coderobe that's good news! The stuttering, however, isn't... Try the debug version and see if it gives predictable output, and see if it ever says "Going to sleep" and "Waking up."

Azkali, 2020-01-23:
Here is my debug log
[kernellog.txt.zip](https://github.com/lazd/VoodooI2CGoodix/files/4105687/kernellog.txt.zip)

lazd, 2020-01-23:
@Azkali looks fine... Can you post another log that includes only an interaction that stuttered, and annotate it so we can see where it stuttered?

Azkali, 2020-01-23:
In fact you're right it does not stutter looking at it twice, but there is definitely something going on, the screen feels much **harder** to draw on, comparing to Linux or Windows.

For example, when I try to drag a big area on the desktop screen, it *seems* to stutter, but in fact it does not try draw a straight line and the *stutter* effect isn't present.

Is there a way the tweak the screen sensitivity in any way ?

Azkali, 2020-01-23:
`kernel_task` looks constant while interacting

lazd, 2020-01-23:
@Azkali can you send a [screen recording](https://support.apple.com/guide/quicktime-player/record-your-screen-qtp97b08e666/mac)? Try dragging on desktop, drawing with a paintbrush in Photoshop, etc.

Azkali, 2020-01-24:
Here you go :
[drag-stutter.mov.zip](https://github.com/lazd/VoodooI2CGoodix/files/4106320/drag-stutter.mov.zip)

lazd, 2020-01-24:
@Azkali and @coderobe that's definitely a clear problem, though I can't quite tell from the logs and video what's going on. Can one of you file a separate issue for this with annotated debug logs and a video so we can address it?

coderobe, 2020-02-11:
The stuttering seems intermittent, can't figure out why it's happening. Either way, that's a different issue - so i think we can close this one and open another one about that.

lazd, 2020-02-12:
Thanks @coderobe, I wonder if it has to do with the speed of the I2C bus or something. Please do go ahead and open another issue for this and let me know if you have any other insights.

commit 1e3ead1e (2020-01-22): feat: add support for GDIX1002 devices, closes #15
closed 2020-02-12 by lazd