lazd/VoodooI2CGoodix #17: 10.15.5 Bug

kostaskas, 2020-06-19:
## Description
double click doesn't work, Gabe clicking 3-4 times will do the job doubler click should, page scroll doesn't even work, sometimes works with very low recognition rate, meaning that page will move only 1cm up/down

## Steps to reproduce

1. Touch something...
2. Do something else...
3. Then this happened...

## Expected behavior

## Screenshot or video

## Environment
 - **Computer**: 
 * **Goodix Touchscreen model**: 
 - **VoodooI2C version:** 
 - **VoodooI2CGoodix version:** 
 - **macOS Version:**

kostaskas, 2020-06-19:
Finally it was VooDooI2c version the I updated, with 2.3 it works ok

lazd, 2020-06-21:
Ok interesting, sounds like we need 2.3 then. Can you send a PR to update the README to say "Make sure you use VoodooI2C 2.3, 2.4 doesn't work" ?

Also, can you test 2.4.0, 2.4.1, etc and see if they fail too?

andot, 2020-06-29:
Computer: GPD P2 Max
VoodooI2C version:2.4.3
VoodooI2CGoodix version:0.2.3
macOS Version: 10.15.5 19F101

doesn't work, too.

kostaskas, 2020-06-29:
> Ok interesting, sounds like we need 2.3 then. Can you send a PR to update the README to say "Make sure you use VoodooI2C 2.3, 2.4 doesn't work" ?
> 
> Also, can you test 2.4.0, 2.4.1, etc and see if they fail too?

I will test and report

kostaskas, 2020-06-29:
> **Computer:** GPD P2 Max (BIOS 0.29)
> **VoodooI2C version:** 2.4.3
> **VoodooI2CGoodix version:** 0.3.1
> **macOS Version:** 10.15.5 19F101
> 
> doesn't work, too.
> 
> VoodooI2C version 2.3, 2.4, 2.4.1, 2.4.2 also doesn't work.

Also 2.3 doesn't work?  Try all files from EFI folder,  it should work as reported

andot, 2020-06-30:
> > **Computer:** GPD P2 Max (BIOS 0.29)
> > **VoodooI2C version:** 2.4.3
> > **VoodooI2CGoodix version:** 0.3.1
> > **macOS Version:** 10.15.5 19F101
> > doesn't work, too.
> > VoodooI2C version 2.3, 2.4, 2.4.1, 2.4.2 also doesn't work.
> 
> Also 2.3 doesn't work? Try all files from EFI folder, it should work as reported

If use this EFI: https://github.com/andot/GPD-P2-MAX-Hackintosh with bios 0.25. VoodooI2C 2.3 works.
But when I update the bios to 0.29 and update the EFI to the latest version of https://github.com/Azkali/GPD-P2-MAX-Hackintosh, the touchscreen can't work, even change the VoodooI2C to 2.3.

jamesfxh, 2020-06-30:
> > > **Computer:** GPD P2 Max (BIOS 0.29)
> > > **VoodooI2C version:** 2.4.3
> > > **VoodooI2CGoodix version:** 0.3.1
> > > **macOS Version:** 10.15.5 19F101
> > > doesn't work, too.
> > > VoodooI2C version 2.3, 2.4, 2.4.1, 2.4.2 also doesn't work.
> > 
> > 
> > Also 2.3 doesn't work? Try all files from EFI folder, it should work as reported
> 
> If use this EFI: https://github.com/andot/GPD-P2-MAX-Hackintosh with bios 0.25. VoodooI2C 2.3 works.
> But when I update the bios to 0.29 and update the EFI to the latest version of https://github.com/Azkali/GPD-P2-MAX-Hackintosh, the touchscreen can't work, even downgrade the VoodooI2C to 2.3.

I found the touch screen code from the DSDT file. After editing it to the SSDT file, 
voodooi2c v2.3 works, but 2.4, 2.4.1, 2.4.2, 2.4.3 doesn't work.

[SSDT-TCSE.aml.zip](https://github.com/lazd/VoodooI2CGoodix/files/4851502/SSDT-TCSE.aml.zip)

andot, 2020-06-30:
@jamesfxh Thank you, voodooi2c v2.3 works now! 👍

lazd, 2020-07-01:
Thanks everyone for reporting back with your experiences. This will be very helpful to others that have the same issue.