lazd/VoodooI2CGoodix #18: Doesn't work on Big Sur 11.0 Beta 4

kostaskas, 2020-08-15:
## Prerequisites

* [ ] I have followed the [VoodooI2C installation instructions](https://voodooi2c.github.io/#Installation/Installation)
* [ ] I have [patched my DSDT](https://github.com/alexandred/VoodooI2C-Patches)
* [ ] I have followed the [GPIO pinning guide](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning)
* [ ] VoodooI2C recognizes my I2C controllers during a verbose boot
* [ ] I have followed the [VoodooI2CGoodix installation instructions](https://github.com/lazd/VoodooI2CGoodix#installation)
* [ ] I have tried tapping my screen and nothing happens

## Environment
 - **Computer**: 
 * **Goodix Touchscreen model**: 
 - **VoodooI2C version:** 
 - **VoodooI2CGoodix version:** 
 - **macOS Version:** 

## `DSDT.zip`

## `IOReg.zip`

## `VoodooI2C.log`

## Additional context

kostaskas, 2020-08-15:
Used OC 0.6.0 but touch screen doesn't work, even with VooDooI2C 2.4.4 version

lazd, 2020-08-17:
Interesting. No logs?

kostaskas, 2020-08-19:
> Interesting. No logs?

VoodooI2C.log?

PR #19 (2024-08-21): fix issue 18
works on big sur

Fixes #18
closed 2024-08-21 by lazd