lazd/VoodooI2CGoodix #21: wired acpi table

anonymous5l, 2021-04-20:
## Prerequisites

* [ ] I have followed the [VoodooI2C installation instructions](https://voodooi2c.github.io/#Installation/Installation)
* [ ] I have [patched my DSDT](https://github.com/alexandred/VoodooI2C-Patches)
* [ ] I have followed the [GPIO pinning guide](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning)
* [x] VoodooI2C recognizes my I2C controllers during a verbose boot
* [x] I have followed the [VoodooI2CGoodix installation instructions](https://github.com/lazd/VoodooI2CGoodix#installation)
* [x] I have tried tapping my screen and nothing happens

## Environment
 - **Computer**: Lattepanda Alpha 864s (DFRobot 7 inch touchscreen)
 * **Goodix Touchscreen model**: GDIX1001
 - **VoodooI2C version:** 2.6.5
 - **VoodooI2CGoodix version:** 4.0 https://github.com/lazd/VoodooI2CGoodix/pull/19
 - **macOS Version:** Big Sur (11.2.3)

## `DSDT.zip`

[DSDT.zip](https://github.com/lazd/VoodooI2CGoodix/files/6345501/DSDT.zip)

## `IOReg.zip`

[MacBook Pro.zip](https://github.com/lazd/VoodooI2CGoodix/files/6345521/MacBook.Pro.zip)

## `VoodooI2C.log`

log show nothing use `dmesg`

```bash
[    2.155944]: VoodooI2CPCIController::pci8086,9d60 Starting I2C controller
[    2.155963]: VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
[    2.155982]: VoodooI2CPCIController::pci8086,9d60 Publishing nub
[    2.158817]: VoodooI2CControllerDriver::pci8086,9d60 Probing controller
[    2.158822]: VoodooI2CControllerDriver::pci8086,9d60 Found valid Synopsys component, continuing with initialisation
[    2.158856]: VoodooI2CControllerNub::pci8086,9d60 SSCN not implemented in ACPI tables
[    2.158862]: VoodooI2CControllerNub::pci8086,9d60 FMCN not implemented in ACPI tables
[    2.158865]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Error getting bus config, using defaults where necessary
[    2.158944]: VoodooI2CControllerDriver::pci8086,9d60 Publishing device nubs
[    2.159899]: VoodooI2CPCIController::pci8086,9d62 Starting I2C controller
[    2.159922]: VoodooI2CPCIController::pci8086,9d62 Set PCI power state D0
[    2.159936]: VoodooI2CPCIController::pci8086,9d62 Publishing nub
[    2.160032]: VoodooI2CControllerDriver::pci8086,9d62 Probing controller
[    2.160038]: VoodooI2CControllerDriver::pci8086,9d62 Found valid Synopsys component, continuing with initialisation
[    2.160066]: VoodooI2CPCIController::pci8086,9d61 Starting I2C controller
[    2.160075]: VoodooI2CControllerNub::pci8086,9d62 SSCN not implemented in ACPI tables
[    2.160081]: VoodooI2CControllerNub::pci8086,9d62 FMCN not implemented in ACPI tables
[    2.160084]: VoodooI2CControllerDriver::pci8086,9d62 Warning: Error getting bus config, using defaults where necessary
[    2.160089]: VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
[    2.160103]: VoodooI2CPCIController::pci8086,9d61 Publishing nub
[    2.160122]: VoodooI2CControllerDriver::pci8086,9d62 Publishing device nubs
[    2.160182]: VoodooI2CControllerDriver::pci8086,9d61 Probing controller
[    2.160187]: VoodooI2CControllerDriver::pci8086,9d61 Found valid Synopsys component, continuing with initialisation
[    2.160237]: VoodooI2CControllerDriver::pci8086,9d61 Got bus configuration values
[    2.160280]: VoodooI2CControllerDriver::pci8086,9d61 Publishing device nubs
[    2.160285]: VoodooI2CControllerDriver::pci8086,9d61 Found I2C device: GDIX1001
[    2.160525]: VoodooI2CDeviceNub::GDIX Got GPIO Controller! VoodooGPIOSunrisePointLP
[    2.662306]: VoodooI2CGoodixTouchDriver::Probing
[    2.683111]: VoodooI2CGoodixTouchDriver::Reading version...
[    2.690025]: VoodooI2CGoodixTouchDriver::ID 911, version: 1060
[    2.691410]: VoodooI2CGoodixTouchDriver::Reading config...
[    2.701823]: VoodooI2CGoodixTouchDriver::xOutputMax = 1024
[    2.705889]: VoodooI2CGoodixTouchDriver::yOutputMax = 600
[    2.707197]: VoodooI2CGoodixTouchDriver::maxTouches = 5
[    2.713920]: VoodooI2CGoodixTouchDriver::screenTouchLevel = 80
[    2.725651]: VoodooI2CGoodixTouchDriver::screenLeaveLevel = 50
[    2.727221]: VoodooI2CGoodixTouchDriver::lowPowerInterval = 3
[    2.734076]: VoodooI2CGoodixTouchDriver::refreshRate = 5
[    2.740944]: VoodooI2CGoodixTouchDriver::xThreshold = 0
[    2.742254]: VoodooI2CGoodixTouchDriver::yThreshold = 0
[    2.748812]: VoodooI2CGoodixTouchDriver::Device initialized
[    2.856617]: VoodooI2CGoodixTouchDriver::VoodooI2CGoodixTouchDriver has started
[    2.856639]: VoodooI2CGoodixEventDriver::Published multitouch interface
[    2.857745]: VoodooI2CGoodixEventDriver::Configuring multitouch interface with dimensions 1024,600 and 11 transducers
```

## Additional context

kext look fine, but the screen no response, no i2c event.

balopez83, 2021-04-20:
@anonymous5l It looks like you might not have followed the instructions listed on the main page. Your DSDT must be patched in order for the kext to work. While yes it is seeing your touch screen, it has no clue what to do with it. Unfortunately, I am at work so can't dig too far into it, but I am pretty sure at first glance your issue is related to a missing section. Here is what I can help with for now to get you in the right direction.

I encourage you to review the I2C GPIO patching instructions located here [https://github.com/VoodooI2C/VoodooI2C/blob/master/Documentation/GPIO%20Pinning.md](url) 
as that will show you exactly what you need to do to get your touch screen running.

Basically your i2cserialbus sections located in the CRS method of your gdix device need to be named SBFG and SBFB and you must have the following in your CRS method:
`            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                Return (ConcatenateResTemplate (SBFB, SBFG))
            }
`

If you are only editing your DSDT and placing that in your EFI (using Clover) then you should be good editing it directly and not adding renames in your config. If you are using (OpenCore) as your bootloader which is recommended, then you must convert the edits to an SSDT and add renames in your config file which is best for long term use for many reason which I won't go into here. Both of these methods will most certainly require the SSDT-XOSI and a rename in the config that renames _OSI to XOSI.

You can use the following SSDT and EFI as an example of some of the edits you may need to get your touch screen working. 
MiniBook Touch SSDT (Goodix Panel) [https://github.com/THEDEVIOUS1/CHUWI-MINIBOOK-HACKINTOSH/blob/master/EFI-OC/EFI/OC/ACPI/SSDT-Touch.aml](url)
MiniBook EFI [https://github.com/THEDEVIOUS1/CHUWI-MINIBOOK-HACKINTOSH](url)
GPD P2 Touch SSDT (Goodix Panel) [https://github.com/Azkali/GPD-P2-MAX-Hackintosh/blob/master/EFI/CLOVER/ACPI/patched/SSDT-TCSE.aml](url)
GPD P2 EFI [https://github.com/Azkali/GPD-P2-MAX-Hackintosh](url)

Its possible you might be able to plug in one of those SSDT's and it might work out of the box BUT you will still need to do the DSDT renaming in the config as demonstrated in the above EFI examples. If you use the SSDTs DO NOT edit your DSDT and place the DSDT in your EFI. You must use renames of the DSDT in your config file only regardless of using Clover or OpenCore when using SSDT's.

I also might suggest editing the DSDT first just to get your stuff in order and then converting your DSDT edits to an SSDT. If you need further help in this please post your DSDT and/or SSDT with edits as well as your config file with any touch edits. Otherwise I won't be able to determine what else you are missing to make it work. It might be easier for you to simply attach your entire EFI BUT make sure you do not include your Themes or Resources folders as they can bloat the EFI considerably. 

Hope this helps a bit.

anonymous5l, 2021-04-21:
@balopez83 thanks so much for help, i made hot patch with GPIO Pin and rename _OSI to XOSI patch `SSDT-XOSI.aml` but still not response. then i found log
 
```bash
[    2.693956]: IOHIDPointingEventDevice:0x10000034b start
[    2.694319]: IOHIDPointingEventDevice:0x10000034b creating interfaces
[   12.883004]: IOHIDPointingEventDevice:0x10000034b open by IOHIDEventDriver 0x100000408 (0x0)
[   13.519186]: DK: AppleUserHIDEventDriver-0x100000478::start(IOHIDInterface-0x100000470) ok
[   13.527934]: DK: AppleUserHIDEventDriver-0x10000047a::start(IOHIDInterface-0x100000476) ok
[   13.544471]: DK: AppleUserHIDEventDriver-0x100000484::start(IOHIDInterface-0x100000483) ok
[   13.562551]: DK: AppleUserHIDEventDriver-0x10000048f::start(IOHIDInterface-0x10000048e) ok
[   20.658190]: IOHIDLibUserClient:0x0 [PerfPowerService] Entitlements 10 privilegedClient : Yes
[   20.658205]: IOHIDLibUserClient:0x100000577 start
[   20.658247]: IOHIDLibUserClient:0x100000577 setValid: from false to true
[   20.658253]: IOHIDLibUserClient:0x100000577 setStateForQueues: 0x0
[   20.658297]: IOHIDLibUserClient:0x100000577 open
[   20.658304]: VoodooInputSimulatorDevice:0x10000033a open by IOHIDLibUserClient 0x100000577 (0x0)
[   20.658313]: IOHIDLibUserClient:0x100000577 setValid: from true to true
[   20.658348]: IOHIDLibUserClient:0x100000577 close
[   20.658351]: VoodooInputSimulatorDevice:0x10000033a close by IOHIDLibUserClient 0x100000577 (0x0)
[   20.658358]: IOHIDLibUserClient:0x100000577 setValid: from true to false
[   20.658360]: IOHIDLibUserClient:0x100000577 setStateForQueues: 0x1
[   20.658494]: IOHIDLibUserClient:0x100000577 stop
[   20.658498]: IOHIDLibUserClient:0x100000577 setStateForQueues: 0x2
[   20.658501]: IOHIDLibUserClient:0x100000577 close
```

seem like the goodix power off can't find why

goodix driver work fine

```bash
[    2.169947]: VoodooI2CGoodixTouchDriver::Probing
[    2.170010]: VoodooI2CGoodixTouchDriver::Reading version...
[    2.170252]: VoodooI2CGoodixTouchDriver::ID 911, version: 1060
[    2.170255]: VoodooI2CGoodixTouchDriver::Reading config...
[    2.173935]: VoodooI2CGoodixTouchDriver::xOutputMax = 1024
[    2.173937]: VoodooI2CGoodixTouchDriver::yOutputMax = 600
[    2.173939]: VoodooI2CGoodixTouchDriver::maxTouches = 5
[    2.173940]: VoodooI2CGoodixTouchDriver::screenTouchLevel = 80
[    2.173942]: VoodooI2CGoodixTouchDriver::screenLeaveLevel = 50
[    2.173944]: VoodooI2CGoodixTouchDriver::lowPowerInterval = 3
[    2.173945]: VoodooI2CGoodixTouchDriver::refreshRate = 5
[    2.173947]: VoodooI2CGoodixTouchDriver::xThreshold = 0
[    2.179156]: VoodooI2CGoodixTouchDriver::yThreshold = 0
[    2.180459]: VoodooI2CGoodixTouchDriver::Device initialized
[    2.288317]: VoodooI2CGoodixTouchDriver::VoodooI2CGoodixTouchDriver has started
[    2.288331]: VoodooI2CGoodixEventDriver::Published multitouch interface
[    2.288402]: VoodooI2CGoodixEventDriver::Configuring multitouch interface with dimensions 1024,600 and 11 transducers
```

anonymous5l, 2021-04-22:
finally i used `VoodooI2C` version `2.3` fix that issues that's so strange.

anonymous5l, 2021-04-22:
source code issues, already fix.
use this kext 
[VoodooI2CGoodix.kext.zip](https://github.com/lazd/VoodooI2CGoodix/files/6357362/VoodooI2CGoodix.kext.zip)
should solve problem

closed 2021-04-22 by anonymous5l

lazd, 2021-04-22:
@anonymous5l please send a pull request with source code changes, do not post a kernel extension for download in the comments.

balopez83, 2021-04-23:
Hi @anonymous5l , glad to hear you got it all sorted out. The Goodix touch panel is a finicky one for sure. The Chuwi minibook had a similar issue requiring the 2.3 i2c kext as well for a bit. There is some differences in the way that version and the latest version work and so makes sense that it worked where the latest did not. The important thing is you got it working :)