lazd/VoodooI2CGoodix #24: Slow speed gestures on my winmax fork

postkevone, 2023-07-14:
Hello.
I have created a fork of this project to support Win Max 2020 hardware:
https://github.com/postkevone/VoodooI2CGoodix-winmax

Everything works fine, but the gestures are very slow.
Is there anything I can modify in order to increase gestures speed?

Thank you.

lazd, 2023-07-14:
Huh, what does a slow gesture look like? Can you post a video?

postkevone, 2023-07-15:
Sure, here's the video: 

https://github.com/lazd/VoodooI2CGoodix/assets/49175327/2372b0e4-ef99-4dec-870b-932069fe6a9f

lazd, 2023-07-17:
Also, your changes seem pretty minor on your fork. I see the show/hide cursor stuff, that's absolutely worth contributing. I wonder if you can check any information from the touchscreen or perhaps PLIST config to determine if the `acpi_device->evaluateObject("_PS0")` is required (also what is that doing? where did it come from? does it hurt other devices?) and maybe we can see if `interrupt_source->enable()` is harmless elsewhere. If so, a PR would be nice so the mainline project has your changes instead of a fork.

postkevone, 2023-07-18:
> Got it, so it's not necessarily _slow_, it's just that a scroll gesture is only scrolling a tiny amount when we expect it to scroll a huge amount... Is this also true for zoom? Rotate? I wonder if there's some scaling issue with the values being sent... You'll need to start adding a ton of `IOLog` statements to see what's really going on with the values being sent, I'd probably [start here](https://github.com/lazd/VoodooI2CGoodix/blob/master/VoodooI2CGoodix/VoodooI2CGoodixEventDriver.cpp#L326-L336).

Yes every multitouch gesture scrolls by a tiny amount. Thank you for your help. I am kind of new to debug stuff of MacOS, do you know any site/guide where I can learn how to debug this kext properly?

postkevone, 2023-07-18:
> Also, your changes seem pretty minor on your fork. I see the show/hide cursor stuff, that's absolutely worth contributing. I wonder if you can check any information from the touchscreen or perhaps [...]

I got the `acpi_device->evaluateObject("_PS0")` line from here: https://github.com/lazd/VoodooI2CGoodix/pull/20/commits/10b4f985e0a620240345b942c0ee0fcdf88e30b4
I also don't know if I really need it or not, but after some testing i notice the kext works fine even with that line.

Regarding `interrupt_source->enable()`, it is necessary for the driver to start on the winmax. Another think that doesn't work properly is wakeup after sleeping, could it be related in some way?

lazd, 2023-07-18:
For debugging, see existing usage of `IOLog` in the code and google how to read logs.

As far as your sleep issues, that’s 100% patching. You’ll never get this working right without correctly patching, and that’s not something I know anything about.