lazd/VoodooI2CGoodix #27: GPD Win 2 on Sequoia 15.4 - GDXI1002 not working

CHR15714N, 2025-04-04:
## Prerequisites

* [x ] I have followed the [VoodooI2C installation instructions](https://voodooi2c.github.io/#Installation/Installation)
* [ x] I have [patched my DSDT](https://github.com/alexandred/VoodooI2C-Patches)
* [ x] I have followed the [GPIO pinning guide](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning)
* [x ] VoodooI2C recognizes my I2C controllers during a verbose boot
* [ x] I have followed the [VoodooI2CGoodix installation instructions](https://github.com/lazd/VoodooI2CGoodix#installation)
* [ ] I have tried tapping my screen and nothing happens

## Environment
 - **Computer**: 
 * **Goodix Touchscreen model**: 
 - **VoodooI2C version:** 
 - **VoodooI2CGoodix version:** 
 - **macOS Version:** 

## `DSDT.zip`

## `IOReg.zip`

## `VoodooI2C.log`

## Additional context
I am trying to get the touchscreen to work with Opencore

 Nearly everything is working and touchscreen is crucial for scrolling... I am no coder, just asking for help on my device.  You can see my post on https://www.tonymacx86.com/threads/macos-15-4-sequoia-on-gpd-win-2-work-in-progress.331747/
Thank you.

[DSDT.aml.zip](https://github.com/user-attachments/files/19605496/DSDT.aml.zip)
[IOreg.zip](https://github.com/user-attachments/files/19605494/IOreg.zip)
[VoodooI2C.log](https://github.com/user-attachments/files/19605495/VoodooI2C.log)
[config.plist.zip](https://github.com/user-attachments/files/19605630/config.plist.zip)

lazd, 2025-04-05:
Looks like it can't find IOGraphicsFamily:

```
2025-04-04 15:33:46.523742+0200 0x27c      Default     0x0                  0      0    kernel: Kext net.lazd.VoodooI2CGoodix - library kext com.apple.iokit.IOGraphicsFamily not found.
2025-04-04 15:33:46.523752+0200 0x27c      Default     0x0                  0      0    kernel: Can't load kext net.lazd.VoodooI2CGoodix - failed to resolve library dependencies.
2025-04-04 15:33:46.523758+0200 0x27c      Default     0x0                  0      0    kernel: Kext net.lazd.VoodooI2CGoodix failed to load (0xdc00800e).
2025-04-04 15:33:46.523763+0200 0x27c      Default     0x0                  0      0    kernel: Failed to load kext net.lazd.VoodooI2CGoodix (error 0xdc00800e).
2025-04-04 15:33:46.523803+0200 0x27c      Default     0x0                  0      0    kernel: Couldn't alloc class "VoodooI2CGoodixTouchDriver"
```

Looks similar to [this issue which has a workaround](https://github.com/VoodooI2C/VoodooI2C/issues/125#issuecomment-461927427). Can you try that?

lazd, 2025-04-05:
This is silly, but can you try this kext?

[VoodooI2CGoodix-0.4.0-plist-edited-removed-IOGraphicsFamily.kext.zip](https://github.com/user-attachments/files/19614376/VoodooI2CGoodix-0.4.0-plist-edited-removed-IOGraphicsFamily.kext.zip)

CHR15714N, 2025-04-05:
Hello and thank you for the support. The kext doesn’t work unfortunately. I also tried deleting that dependency before. As for the solution you linked, I cant find a section <key>KernelAndKextPatches</key> in my plist. I cant quite understand where i should put that code…

CHR15714N, 2025-04-05:
Also, i can actually see the device in IOreg under I2C1 controller identified as TPL1.

lazd, 2025-04-05:
> Also, i can actually see the device in IOreg under I2C1 controller identified as TPL1.

Are you sure that's the latest log?

lazd, 2025-04-06:
Please try on and older version of macOS and let me know if it works with your hardware, that way we can see if it's just Sequoia or if it's your hardware.

CHR15714N, 2025-04-06:
> Please try on and older version of macOS and let me know if it works with your hardware, that way we can see if it's just Sequoia or if it's your hardware.

Ok, I will try on Monterey

CHR15714N, 2025-04-06:
Same behavior on Monterey :( (I can see the device IOreg as TPL1)

[GPD.ioreg.zip](https://github.com/user-attachments/files/19622863/GPD.ioreg.zip)
[VoodooI2C.log](https://github.com/user-attachments/files/19622862/VoodooI2C.log)

lazd, 2025-04-07:
That commit was already rolled in, and I have this successfully working all the way up to Sonoma, so that isn't the problem... Honestly not sure what to tell you, I don't have the device here so I can't help debug. Perhaps check with others that have successfully hacked the WIn2

lazd, 2025-04-07:
Good points @balopez83!

@CHR15714N, you could try using the exact VoodooI2C* kexts from this EFI, which is known good working with Goodix touchscreens on Sonoma: https://github.com/lazd/LattePanda2Alpha864sOpenCoreHackintosh/tree/main/EFI/OC/Kexts

CHR15714N, 2025-04-07:
@lazd I’m on it!

CHR15714N, 2025-04-07:
@lazd @balopez83 IT WORKS! IT WORKS! IT WORKS!
At least on Monterey!
I dont know if multitouch gestures are supposed to work, but i can at least move the cursor! After a week trying, thank you so much! I will try getting back on Sequoia and see if it still works.

lazd, 2025-04-08:
@CHR15714N glad to hear it! Looks like that kext was the magic. I'll have to release a new version with those changes. I think multitouch just works out of the box, try 2 finger, 3 finger, etc.

I can't remember how I set that kext up, so can you answer:

1. Does tap work?
2. Does right click (tap and hold) work?
3. Does scrolling work? Do you have to use 2 fingers or does it scroll with 1 finger?

CHR15714N, 2025-04-08:
@lazd 
1) i can literally only move the cursor, which means the cursor teleports under my finger and does 1 click where i tap. I can doubletap an icon to open it.
2) tap and hold does nothing
3) scrolling works with 1 finger. I think it is clicking and dragging down.
4) When i tap with 2 fingers, the cursor simply teleports in the midpoint between them.
5) When I tap and hold with 2 fingers, using a third finger causes strange random behaviors. The cursor goes generally to random positions, but sometimes I am able to move the cursor on the screen by swiping with the third finger, without having it teleport under the finger. It’s a bit like using a mouse.

CHR15714N, 2025-04-08:
@lazd I made another interesting discovery. The magic isn’t contained in the older version of Vodooi2c. What makes my touchscreen work is simply the Voodooi2cgoodix kext from that lattepanda build. I can also use it with the latest version of Voodooi2c.
As for Voodooi2cHID I don’t know of I really need this kext. Loading it or deleting it, makes no difference.

CHR15714N, 2025-04-08:
Here is the latest Voodooi2c log.

[VoodooI2C.log](https://github.com/user-attachments/files/19644245/VoodooI2C.log)

CHR15714N, 2025-04-08:
Ok, after some testing i have come to these conclusions: I am going to stick to the LattePanda Voodooi2c because it allows me to do more things. With the latest voodooi2c i can just move the mouse.
With the lattepanda version multitouch actually works and i can:
- Scroll (1 finger): scrolling is inverted, to use natural scrolling i have to actually deactivate the option
- pinch to zoom
- rotate gesture
- close 4 fingers for launchpad
- swipe up with 3 fingers for mission control (strangely this works normally, not inverted)
- swipe down with 3 fingers for app expose (strangely this works normally, not inverted)

Strange behaviors 
- swiping with 2 fingers from left or right gives reverted results, for example to have control center (right side of the screen) i have to swipe from the left edge. Same for safari pages swiping left or right. This may be because the display is natively in potrait and i am using a 27- degree rotation.

What i can’t do:
- move the cursor like on a real trackpad (cursor is alwways teleporting under my finger)
- right click (i can do a sort of right click tapping with 2 fingers but the cursor lands randomly between them)

Here the last log:

[VoodooI2C.log](https://github.com/user-attachments/files/19650464/VoodooI2C.log)

Overall I would say, this was a success! I am happy with the touch screen, thank you so much for the support!

closed 2025-04-08 by lazd

CHR15714N, 2025-04-08:
@lazd Thank you. No, unfortunately long press doesn´t right click.